keirf/greaseweazle #502: Issue with latest python release (I think)

sdsnatcher73, 2024-10-23:
On my MacBook with python 3.13 installed through brew I keep running into issues when GreaseWeazle tries to read or verify written data:

Format pc98.2hd
Writing c=0-76:h=0-1
T0.0: Writing Track (Flux: 166.7ms period, 183.3 ms total, Terminate at index)
** FATAL ERROR:
**'bitarray.bitarray' object has no attribute 'itersearch'**

I think at least it is related to the python version, going back to earlier GW releases did not fix it while those worked before.

keirf, 2024-10-23:
Either I need to require >=3 and fix the code, or I need to require <3. I guess the latter is simpler for now. Let me see.

keirf, 2024-10-23:
Should be fixed in release tags `v1.21` and `latest`.

closed 2024-10-23 by keirf

branchonequal, 2024-12-29:
I "fixed" it locally by replacing all occurrences of `itersearch()` with `search()` (according to the [bitarray version 3 transition document](https://github.com/ilanschnell/bitarray/blob/master/doc/bitarray3.rst)) which seems to work (I could at least write a disk without issues).

keirf, 2025-02-11:
On v1.21 you should be able to uninstall the bitarray package and then greaseweazle installation should pull in the required earlier version.