keirf/greaseweazle #506: Memotech diskdef ?!

gilmod714, 2024-11-10:
hello,

is there support in diskdefs for reading Memotech MTX512 floppy disks ?!

I have tried adding the geometry , but it didnt work ..

Thank you,
Gilmo

keirf, 2024-11-10:
How does it fail? Can you take a raw dump (gw read with no --format to a raw flux format such as scp).

gilmod714, 2024-11-10:
ok , I will upload some screen shots now ..

gilmod714, 2024-11-10:
Ok, Thanks Keirf

If I read SCP - unspecified format .. it read the image .. but the resulting image writing it back doesn't work on MTX512

If I do "Memotech type 7"  read , I get a bunch of errors .. (pic #3)

[image]

[image]

[image]

keirf, 2024-11-11:
Can you share the SCP image file? Then I can inspect it.

gilmod714, 2024-11-11:
of course .. please see attached .. this is mostly a blank MTX512 disk with only one file on it called "sdxboot.bas"

the first 2 tracks are CPM system bytes ..

[mydisk.zip](https://github.com/user-attachments/files/17704440/mydisk.zip)

keirf, 2024-11-11:
Load that SCP into HxC software and look at tracks in "stream view" and you will see the data is very noisy. Possibly your drive supports double- and high-density, selected by pin 2. So try adding `--dd L` to your `gw read` command line(s). I think your disk format specifier is probably correct.

gilmod714, 2024-11-11:
Ok , yes I do believe that my drive supports both DD/HD , it's a 3.5" Sony MPF920

I will try what you have suggested,

Thank you Keir

keirf, 2024-11-11:
Oh 3.5-inch drives usually auto-detect the density. They don't take a mode select on pin 2.

It is weird how your dump is worse on the outer (lower-numbered) tracks. So nonetheless it still feels like it could be density related. Suffice to say it is no wonder the software can't retrieve sectors reliably, anyway.

gilmod714, 2024-11-11:
ok, so I changed the floppy to another unit I have "Panasonic JU-256A" .. and I am not getting any errors reading/writing anymore (so I guess Memotech type07 diskdef is correct ... I was able to write a ready-made .hfe image and it works on the MTX side ... however, reading the same disk back and comparing the 2 images in hex-editor , there are some differences in bytes ... 

I am using HD floppies by the way with hole covered , since I don't have any DD disks , if that's makes a difference..

keirf, 2024-11-11:
HD disks with hole covered are usually reliable enough. What disk image type are you comparing in the hex editor?

gilmod714, 2024-11-11:
i am writing an .hfe image ready-made found online-> type 07 , testing it on real MTX512 disk drive works ok, then reading back that same disk I created as .hfe also type07 ..then comparing it against the original .hfe image  .. some bytes are different ..

[image]

[disk-comparison.zip](https://github.com/user-attachments/files/17705598/disk-comparison.zip)

keirf, 2024-11-11:
You can't hex-compare HFE image files. They are raw, containing lower-level info than just sector data, and it can vary across dumps etc. So, case closed!

closed 2024-11-11 by keirf