keirf/greaseweazle #508: Question: Use GreaseWeasle as a pass-through USB to FDD controller in DOS?

Smeedy, 2024-11-16:
Hi,

Been reading up on the project to see if the GW can replace the shoddy FDD 1306 USB controller you see popping up on Ali and Amazon, which only support a limit set of [formats](https://wiki.foone.org/w/Floppy_Controllers) and apparently has all kinds of issues if you search further. Tried to get an answer from documentation, but I think my use case is not so obvious.

Currently I am searching a way to connect a Gotek in a genuine DOS in modern hardware [LLama ITX](https://github.com/eivindbohler/itxllama) Github built project around the Vortex86EX SOM, but that System on Module does not have FDD in the spec. And the only way to connect a genuine floppy drive is through USB.

Using the GreaseWeasle as a pass-through to a real drive - or to a Gotek for that matter - using a USB DOS driver (OS or through bios even) might kill 2 birds with one stone: We would have a capable controller to the system reading the old disks, but also have a way on our daily driver machines to archive our old disks and use it as it was intended.

Can I have your thoughts on this? Is there a way to use the GW as a USB2FDD pass-through?

thanks,
Martijn

Smeedy, 2024-11-19:
Just got my own GW and it's a serial /dev/cu.usbmodem device which is also reflected in the import serial when I go through the source. Which in its own set this in a different perspective from a DOS driver/communication perspective. But also opens possibilities as I can do 115k2 - though I have to bypass the usb2serial, or bake that in the driver as well.

Has something like using the GW as a pass-through ever been considered? What would be the impossibilities?

keirf, 2024-11-19:
FDD USB is a separate protocol unsupported by Greaseweazle. I'm not sure how flexible that protocol actually is. In practice it's only really used for 720k and 1m44 formatted disks.

Smeedy, 2024-11-19:
Exactly. I think I'll just need 2 stones for my 2 birds. I'll got myself that 1306 USB controller thingy and see how far I can take it with different images and formats on the Gotek when it arrives. Let's close this ticket and I can always open a new one when I decide to go down the rabbit hole and start writing dos device drivers and use the API through serial. I am considering it as a project as I've learned a thing or two since my DOS Days decades ago - but not soon.

Loving the GW btw, already been busy dumping my old 3.5" disks from many moons before. Amazing how little errors I encountered. I have been cursing myself ever since I threw out all my 5.25" disks  in a far past. Thanks for your work, got my GW from https://www.retro-updates.com

closed 2024-11-19 by Smeedy

stuaxocabinetoffice, 2024-11-27:
@Smeedy maybe at some point in the future we will have a device that has space to support the different formats, by passing some command so it can present itself as either kind of device.

Smeedy, 2024-11-27:
Do you mean space from a FW perspective and adding necessary calls?

I'm was thinking teaming up with an old DOS programmer I know and look at some open source project we can butcher as a starting point for driver. Splitting up concerns you could start writing a DOS driver from the FDD controller perspective as the interface should be well documented. Intercepting some 13h calls and diverting it to a serial interface with would be the first step, mapping it to GW calls would be the next. Which in turn would pass the call ito a physical drive - be it some standard 3.5/5.25 drive of the DOS area.

Is this where this would not fly since you would need to adhere to specific FDD calls for this pass-through operation mode, which cannot be answered by the GW at this point in time?

stuaxocabinetoffice, 2024-12-03:
I reckon the FDD call would be different, but implementing an INT13h driver to speak to the greaseweazle is definitely the way to go.

There are already some USB drivers for DOS + that might be the place to start.  Since the greaseweazle (and other non-PC floppy drives) can do some things that PC floppy drive controllers couldn't it may be worth adding some extra calls, in that way it would be possible to do things beyond using it to read DOS floppies.

stuaxo, 2024-12-03:
(Swapping to my non-work account I meant to use in the first place).

I realised this is ambiguous:

> I reckon the FDD call would be different, but implementing an INT13h driver to speak to the greaseweazle is definitely the way to go.

In that bit I meant non greaseweazle USB generic floppy drives would have a different way of being called than the greaseweazles - which should in fact pretty straightforward.