keirf/greaseweazle #513: Panasonic/Matsushita JU-257A72P

dlundqvist, 2024-12-13:
Hi,

Have an Panasonic/Matsushita JU-257A72P drive I pulled from a Roland MC-80 (IIRC) that I cannot get to work with GW (latest firmware). I've tried writing both DD and HD floppies with it but it writing always fail verify:

```
neptune ~% gw write --drive b --format ibm.720 ~/gotek/720DD.IMG
Format ibm.720
Writing c=0-79:h=0-1
T0.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T0.0: Writing Track (Verify Failure: Retry #1)
T0.0: Writing Track (Verify Failure: Retry #2)
T0.0: Writing Track (Verify Failure: Retry #3)
** FATAL ERROR:
Failed to verify Track 0.0
neptune ~% gw write --drive b --format ibm.1440 ~/gotek/1440HD.IMG 
Format ibm.1440
Writing c=0-79:h=0-1
T0.0: Writing Track (Flux: 200.0ms period, 220.0 ms total, Terminate at index)
T0.0: Writing Track (Verify Failure: Retry #1)
T0.0: Writing Track (Verify Failure: Retry #2)
T0.0: Writing Track (Verify Failure: Retry #3)
Unknown mark f9
** FATAL ERROR:
Failed to verify Track 0.0
```

Reading is not much better:
```neptune ~% gw read --drive b --format ibm.1440 disk.img
Reading c=0-79:h=0-1 revs=2
Format ibm.1440
Unknown mark fc
Unknown mark d2
Unknown mark f9
Unknown mark fa
Unknown mark f9
Unknown mark f9
T0.0: IBM MFM (12/18 sectors) from Raw Flux (151304 flux in 400.47ms)
Unknown mark f9
Unknown mark f9
T0.0: IBM MFM (13/18 sectors) from Raw Flux (227098 flux in 600.43ms) (Retry #1.1)
Unknown mark f9
Unknown mark f9
Unknown mark be
T0.0: IBM MFM (13/18 sectors) from Raw Flux (228171 flux in 600.41ms) (Retry #1.2)
```

That floppy was just written to another drive, Mitsubishi MF355F-3252MG, by GW.

I'm quite new to different kinds of drives and GW, and I can't find much information about the drive online so not sure what I can do to troubleshoot or fix the issue. Any ideas what could be wrong?

GW info:
```
neptune ~% gw delays
Select Delay: 10us
Step Delay:   10000us
Settle Time:  15ms
Motor Delay:  750ms
Watchdog:     10000ms
Pre-Write:    100us
Post-Write:   1000us
Index Mask:   200us
neptune ~% gw bandwidth

                   Min.   /   Mean   /   Max.   
Write Bandwidth:    9.004 /    9.061 /    9.102 Mbps
Read Bandwidth:     9.214 /    9.215 /    9.228 Mbps

Estimated Consistent Min. Bandwidth: 8.104 Mbps
 -> Max. Flux Rate: 0.912 Msamples/sec
 -> Min. Ave. Flux: 1.097 us
neptune ~% gw info     
Host Tools: 1.21
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4.1
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.6
  Serial:   GWB0BD79A75976C01007189705
  USB:      Full Speed (12 Mbit/s), 128kB Buffer
```

rmueller83, 2025-01-08:
Did you try an external power supply? My JU-257A604P requires 12V which most GW boards do not provide.

keirf, 2025-01-08:
My guess would be misalignment of at least one of the drives involved.

dlundqvist, 2025-01-08:
> Did you try an external power supply? My JU-257A604P requires 12V which most GW boards do not provide.

I have not. I hear the drive working but perhaps it still needs 12V. Ordered one molex PSU and I'll give that a try.

dlundqvist, 2025-01-08:
> I suspect the drive simpy needs repair.

That's very likely, don't think I tried the drive when it was in the Roland equipment unfortunately. I'll start with giving it 12V and take it from there.

dlundqvist, 2025-01-26:
As suspected, 12V did not help. Any recommendations for a guide for repairing floppy drives? I'm new to repairing electronics but keen to learn more.

keirf, 2025-04-19:
Unfortunately it could be all sorts of things. One of the easiest to fix would be dirty heads.