keirf/greaseweazle #515: Successfully read raw file convertion errors

sugix8, 2024-12-23:
Hi Keirf,

I have a old mac 800 3.5 disk which cannot be read fully on original mac.  Thus, I have decided to try saving track by track by different parameters so I can combine all the raw files (--raw) to create a full working disk - this required me to use multiple floppy drives from multiple manufactors.  After repeating the procedure so many times, I have finally collected all tracks - i.e. gw would stop reading from the track and finish the process with report 'n/n sectors read, 100%' completion..

I have put all files under a single folder and tried the following command:
"C:\\greaseweazle-1.21-win64\greaseweazle-1.21\gw.exe" convert --format mac.800 "C:\\rawtracks00.0.raw" "C:\\ima-ge.img"

The output resulted in bunch of missing sectors on various tracks. 
I am assuming the converter is giving up without reading throughout the file?  Perhaps converting from last flux data and backwards may be a solution?  On few of the most troubled tracks I used parameters:1 revision, 10 retires, 10 seeks. 
Kindly advise how I should approach to decode from the raw data I have saved..

(And thanks for the asesome software!)

keirf, 2024-12-23:
So AIUI you gave used track-per-file Kryoflux stream format, and for each track M.N you have a 100% good read called `rawtracksM.N.raw`? Or at least what *should* be 100% decodeable.

My guess is that this is because PLL parameters get reset on each retry during `gw read`. But then all retry fluxes get concatenated together in the output file and decoded by `gw convert` in one single pass, with no corresponding reset of PLL parameters at the retry splice points. Hence the PLL/decode behaviour during convert is *not* the same as during read.

I would also guess that this will be decodeable still with a suitable choice of PLL parameters. If you can share your raw files I can take a look.

sugix8, 2024-12-23:
Hi Keirf,

Thanks for the prompt reply and explanation.  Your undrestanding is correct regarding what I tried to describe ;)
I have put all the raw files with conversion log file I pasted on google drive.  Is there a way I can contact you directly?

sugix8, 2024-12-23:
> I will take a look. These flux files are quite big which makes manipulating them rather slow.

Hi Keir,Thank you very much for your attention to this issue.Yes, it’s very large due to my retry settings.I must have re-read track zero a few thousands of times to get the first 100%.BTW, do you know if MFS have a checksum sort of mechanism to ensure what is read from the sector is error free?Please take a look when you get a chance.Thank you!Sugi, Tokyo--------------YS

keirf, 2024-12-24:
I'm not sure there is a higher-level filesystem checksum, only the Mac GCR block checksums. Those are moderately weak, albeit 24 bits compared with IBM 16-bit CRC.

HxC tools choke on the large files, so to inspect say track 0, I convert some revs to SCP:
```
gw convert --tracks=c=0 egb00.0.raw egb.scp
```

Now load into HxC software, Track Analyzer, Stream View, and Inspect cyl 0, head 0. Note on every rev there is a disruption in the horizontal bit bands. The two lower bands merge together here. Zoom in and you can see many bitcells. The output here from the PLL will be essentially random. So what you are doing is retrying until you get "lucky" and pass the checksum. It also explains why it is sensitive to `gw read` vs `gw convert` as you are using the PLL as a RNG, and it's going to be sensitive to starting conditions as to what random crap it spits out.

You need to inspect the disk and hopefully giving it a clean may be sufficient.

closed 2024-12-30 by keirf

sugix8, 2024-12-31:
Hi Keirf,

(Sorry for the late reply as I thought I would get a email notice when updated)
Thanks for the inspection and analysis. 
I did try reading on HxC Software and yes it did choke on large raw tracks.  Let me try the SCP conversion methods you have explained.

I did a full cleaning of the floppy surface to a level that cannot be done any further.

Could you advise a command option to output a binary data file per track (when successfully decoded after retries)?  I would like to inspect the binary data when (it is thought to be successfully) read.