keirf/greaseweazle #518: Trying to convert IBM XDF (Extended Density Format)

ceneblock, 2025-01-28:
Hello,
  I'm trying to image a copy of PC DOS 7.0 on floppy. Disks 2-5 are listed as the "XDF" format (which appears to be different than Sharp X68000 (#238).

Using HxC, I'm given the following image

[image]

I think I've analyzed it correctly; however, there smaller sectors on the outermost track? I've highlighted two below, but they surround the entire disk. 

[image]

I'm happy to contribute my findings; however, I don't know how I'd write the config file for differently sized sectors.

Additional information:
Greaseweazle created based on bluepill

gw info
```
Host Tools: 1.21
Device:
  Port:     /dev/ttyACM1
  Model:    Greaseweazle F1
  MCU:      72MHz, 10kB SRAM
  Firmware: 1.6
  Serial:   [Removed On Purpose]
  USB:      Full Speed (12 Mbit/s), 4kB Buffer
```

./hxcfe --help
```
HxC Floppy Emulator : Floppy image file converter v2.16.10.1
```

host info:

uname -a
```
Linux main 6.12.8-1-default #1 SMP PREEMPT_DYNAMIC Thu Jan  2 12:47:33 UTC 2025 (ab2dfe6) x86_64 x86_64 x86_64 GNU/Linux
```

gcc --version
```
gcc (SUSE Linux) 14.2.1 20241007 [revision 4af44f2cf7d281f3e4f3957efce10e8b2ccb2ad3]
Copyright (C) 2024 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

andrewcleveland, 2025-02-02:
I forked a new codec and image format to read a set of PC DOS 7 install floppies, but they have bad sectors so I wasn't able to verify if it works completely.

https://github.com/keirf/greaseweazle/commit/5344f6ac7f28da23c625098685d57dc4e5262542

`gw read --format ibm.xdf test.ibmxdf`

The ibmxdf codec reads both the "stub" image and the actual XDF image consecutively.
The ibmxdf image format omits the stub image, but you can do a read with ibm.1440 and `--tracks c=0,h=0` to get it.

I followed this description of the IBM XDF format.
https://www.os2museum.com/wp/the-xdf-diskette-format/

keirf, 2025-02-03:
@andrewcleveland Thanks very much for this -- it's very rare to get a code contribution let alone a modified codec!

However, I added IBMXDF to FlashFloppy some years ago, and my recollection is that the layout is considerably more moulinexed than your implementation. It's (tersely!!) described here: https://github.com/keirf/flashfloppy/blob/master/src/image/img.c#L1373

That work was based on the best open source implementation I could find, in fdutils (https://github.com/lipro-cpm4l/fdutils/blob/cpm4l/fdutils-5.5-20081027/src/xdfcopy.c) and also with reference to raw dumps of original XDF disks.

Your implementation looks to have the wrong layout for sector IDs, and have the wrong data in those sectors. ;) That said, I do wonder whether broadly speaking your split into codec changes plus a new image handler is good -- a *lot* of the moulinexing could perhaps cleanly be handled in the image handler, which simply needs beefing up (and it needs to be able to *import* XDF images also). Perhaps with reference to FlashFloppy and fdutils code. 

FlashFloppy/Gotek is of course also:
(a) useful for cross testing and comparison
(b) gives another option for installing on old PCs from XDF images (which is by far the most common reason to go near IBMXDF these days).