keirf/greaseweazle #522: GetFluxStatus: No Index

FuzzyLapin, 2025-02-15:
Hello,

I recently got a Greaseweazle and I'm receiving a 'GetFluxStatus: No Index' error when trying to perform most actions (e.g rpm, read, etc).

Some info and results from my fiddling which may help:
-Running on Linux, GW is on the latest firmware.
-The drive is a 3.5" Sony MPF920.
-Connected with a straight, 34-pin ribbon cable with the red strip facing away from the power connector.
-Powered using the 4-pin connector on the Greaseweazle board. From everything I've read, the GW alone should be enough to power this drive.
-The drive shows a solid green activity light at all times. It's silent when no disk is inserted, but makes a persistent, quiet clicking noise when one is inserted.
-I've tried all possible variations for --drive and get the same error.
-Using gw seek 0 appears to do something (no errors), but using seek with any number above 0 results in 'FATAL ERROR: Track0 signal asserted after seek to cylinder X'. This is while a disk is inserted.
-I've tried multiple disks and they all give the same error.
-gw info displays the following:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.6
  Serial:   GW20BDB92620C18000079C1517
  USB:      Full Speed (12 Mbit/s), 128kB Buffer

I'm new to all this, so anything you can share to help me troubleshoot is appreciated.

keirf, 2025-02-15:
The ribbon cable is wrong way round. Flip it at either the drive or the Greaseweazle (but not both!).

FuzzyLapin, 2025-02-20:
Thanks! Sorry for taking a while to get back to you, but that was indeed the issue.

closed 2025-02-20 by keirf