keirf/greaseweazle #523: hot ic

tecfixed, 2025-02-17:
Hi,

I  have killed the AT32F ic on board, i dropped the powered board and it landed on the chassis of the floppy drive, now all that happens is usb over current warning appears on windows.

Ive pulled the regulator off and fed 1.5V at 500mA into the 3.3v header and the ic is getting hot so ive pulled the ic and short is gone.
Do you have a list of compatible replacements and is reprogramming them straightforward, was just getting started dumping disks nicely, knew i should have printed the case!!

thanks,
Paul.

keirf, 2025-02-18:
AT32F403A, AT32F403, AT32F415, in descending order of preference. In LQFP48 package.

tecfixed, 2025-02-18:
Thanks for that, I've ordered some AT32F415 from lcsc and printing a case now so it doesn't happen again!!

closed 2025-03-02 by keirf

tecfixed, 2025-03-04:
just to confirm, new ic fitted, programmed and working again.