keirf/greaseweazle #525: greaseweazle v4 suddenly do no longer work, USB disconnect

ggaussling, 2025-03-01:
Hi,

I successfuly worked with my greaseweazle v4. But yesterday it stopped working. Maybe due to an ubuntu upgrade?

It only appears the red LED for power I guess.
Sometinmes I get an output in dmesg, often not at all while pulling in and out the USB cable.

`[  438.625006] usb 2-1.5: new full-speed USB device number 5 using ehci-pci
[  438.707431] usb 2-1.5: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
[  438.707439] usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  438.707442] usb 2-1.5: Product: Greaseweazle
[  438.707443] usb 2-1.5: Manufacturer: Keir Fraser
[  438.707445] usb 2-1.5: SerialNumber: GW0045C5947048000007A9870F
[  438.787178] cdc_acm 2-1.5:1.0: ttyACM0: USB ACM device
[  438.787274] usbcore: registered new interface driver cdc_acm
[  438.787276] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[  439.733424] usb 2-1.5: USB disconnect, device number 5
ls: Zugriff auf '/dev/ttyACM0' nicht möglich: Datei oder Verzeichnis nicht gefunden
Host Tools: 1.21
Device:
  Not found
`

keirf, 2025-03-01:
Looks like it is not connecting at all, or connecting only momentarily? Could be USB port or USB cable, or the Greaseweazle itself.

ggaussling, 2025-03-02:
It seems like the communication issue or reset of the device happens on windows too. Sp yes it might be a faulty board. I got this on two USB cables and the power cable is tested fine.

Only the red power led is on and gives a steady but somewhat faint light.

Do you think I can manage to find the culprit? Maybe a capacitor or a solder joint? 

I'm a noob and absolute beginner, but I bought some stuff to get old computers maybe repaired, as I once recapped a monitor with a black screen successfully.

I bought a USB scope, an air soldering kit, a logic tester and xgecu eprom programmer etc. to learn something about computer electronics.

Do you think it might could be helpful to reflow the solder with the air gun?

keirf, 2025-03-02:
I would suggest to test the board with no drive connected (and hence no power cable). This will remove possibility of excessive power draw by the drive.

ggaussling, 2025-03-02:
That doesn't change anything. I misunderstood the power connector though. Does it harm, if the cable is under power instead of delivering power from the board? I guess not, but ...

ggaussling, 2025-03-02:
I ran it this way for almost two years flawlessy. But good to know anyway. U4 is ~37°C, measured with flir ir.  The GWv4 doesn't  appear with the jumper on the 3V3 (first vlr) header. I measured 5V (6th) with GND (2nd) 4.92V (@USB 4.8V or 4.66 at a USB hub). Also 4.92V at the power out. Between the up pin of 3V3 and GND header it has 1,68V. 

I'm not experienced at all, but didn't find anything interesting in the solder joints under my microscope.

Unfortunately nothing appears with 3V3 jumper on. But currently I also don't get any dmesg with the former jumper setting. It apears to be quite dead, while the red led is shining.
```
~$ sudo dmesg | egrep -v '] audit: '| tail -n40 && LANG=C ls /dev/ttyACM0 ; gw --bt info ; sudo lsusb | grep -iv hub
[...]
[   88.908677] kauditd_printk_skb: 1 callbacks suppressed
ls: cannot access '/dev/ttyACM0': No such file or directory
Host Tools: 1.21
Device:
  Not found
Bus 001 Device 003: ID 046d:c077 Logitech, Inc. Mouse
Bus 002 Device 003: ID 05e3:0723 Genesys Logic, Inc. GL827L SD/MMC/MS Flash Card Reader
Bus 004 Device 002: ID 1058:25ee Western Digital Technologies, Inc. My Book 25EE
```

keirf, 2025-03-02:
It could also be the reg itself. If any other part we're shorting I think you'd see it being hotter.

ggaussling, 2025-03-02:
With reg do you mean U4 LD1117S33TR ? Because of the low voltage (in 3.22 and out 1.7V). I'm located in Koblenz Germany. I would think the voltage regulator itself or surrounding capacitors like c13, c12, c18 might be the culprit?

keirf, 2025-03-02:
Yes, U4. It is strange you have only 3.22V at the U4 input. Yet 4.92V at the 5V pin of the jumper block? Those points should be connected with a wire!

Well in this case I would suspect U4.

Anyway, I could send you a GW v4.1. You can find my email in my GitHub profile.

keirf, 2025-03-02:
Perhaps you could also send or attach a close-up picture of the GW v4 including MCU from which I can determine it's one of my stock (and/or let me know where you bought it from).

ggaussling, 2025-03-02:
What are the advandages of GW v4.1 over v4? I'm working with PC, Amiga, AppleII and maybe other formats too. 3.5" and 5,25". Also C64 flippy disks.

On that PCB is printed "Engineered by George R. Mezzomo 2021".  Like this one https://www.tindie.com/products/emmybear/greaseweazle-v4-old-style/, but I think I bought it from [AMIGAstore.eu](http://amigastore.eu/)

I also got a M-Tester LCR T7, but I think I cannot test that LD1117 with it, but maybe the surrounding capacitors.

keirf, 2025-03-02:
Download the schematic from Design Files section of the wiki and you can try removing caps on the 5v side of the circuit. They are only decoupling/bypass caps -- the board can work without them. See if that improves the measured voltage. If not, remove U4. Check voltage at the footprint of the V_in pin of U4. It should certainly then read near 5v. If removing U4 is the fix, U4 needs to be replaced. 

I supply Amigastore.eu, and I'm happy to provide warranty for the very few units I have reported bad. So, if you want, email me and I will arrange to send you a v4.1 unit at no cost to you.

closed 2025-03-02 by keirf

ggaussling, 2025-03-02:
Hello Keirf,

Thank you for your service. I'll send you a message by email for RMA. I appreciate your offer to help me with my case.

Here, what I tried 

LCR T7 measurement 
C12 1163nF esr 2,3Ω
C13 14,2μF esr 0,24Ω
C10 13,47μF esr 0,26Ω

BoM
C12 1μ
C13 10μ
C10 10μ

Without C12, C13, C10 the voltage at the reg is normal (see below), but 3V3 header is still 1,8V.

Measurement voltage
LD1117S33TR
       _
    _|  |-3  
o |_|  |-2
      |_|-1

3 5,0V+
2 3,3V+
1 gnd

Because the 3v3 header has still 1,8V I guess, the device still doesn't show up. 

Most of the resistors seem to be fine. But I didn't measure all of them.

I don't know the tolerances though.

ggaussling, 2025-03-07:
Hi, I swapped the LD1117STR with a new one, and it detects now the debvice.
```
~$ sudo dmesg | egrep -v 'sauid=102|] audit: '| tail -n8 && LANG=C ls /dev/ttyACM0 ; gw --bt info ; sudo lsusb |egrep -iv hub
[426378.311524] kauditd_printk_skb: 31 callbacks suppressed
[426554.468572] usb 2-1.5: new full-speed USB device number 9 using ehci-pci
[426554.548990] usb 2-1.5: New USB device found, idVendor=1209, idProduct=4d69, bcdDevice= 1.00
[426554.548998] usb 2-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[426554.549001] usb 2-1.5: Product: Greaseweazle
[426554.549004] usb 2-1.5: Manufacturer: Keir Fraser
[426554.549006] usb 2-1.5: SerialNumber: GW0045C5947048000007A9870F
[426554.549787] cdc_acm 2-1.5:1.0: ttyACM0: USB ACM device
/dev/ttyACM0
Host Tools: 1.21
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.6
  Serial:   GW0045C5947048000007A9870F
  USB:      Full Speed (12 Mbit/s), 128kB Buffer
Bus 001 Device 003: ID 046d:c077 Logitech, Inc. Mouse
Bus 002 Device 003: ID 05e3:0723 Genesys Logic, Inc. GL827L SD/MMC/MS Flash Card Reader
Bus 002 Device 009: ID 1209:4d69 Generic Greaseweazle
Bus 003 Device 015: ID 054c:002c Sony Corp. USB Floppy Disk Drive
Bus 004 Device 005: ID 1058:25ee Western Digital Technologies, Inc. My Book 25EE

```

Unfortunately I flipped one of the capacitors away with to much pressure with the tweezer and so C12 isn't populated.

All my recent reading tries are claiming Track 0 isn't readable. I think this might be due to the lack of the capacitor, or also the other two capacitors might no longer be sane.

```
~$ gw read --retries 10 floppy.adf && xdftool floppy.adf list && rm floppy.adf
Reading c=0-79:h=0-1 revs=1.1
Format amiga.amigados
Command Failed: GetFluxStatus: No Index
'list' IOError: image file not found
```

```
~$ gw read MyAmigaDisk.scp --raw
Reading c=0-81:h=0-1 revs=3
Command Failed: Seek: Track 0 not found
```

```
~$ gw read --retries 10 --format=ibm.scan floppy.ima && fatcat -l /  floppy.ima && rm floppy.ima
Reading c=0-79:h=0-1 revs=2
Format ibm.scan
Command Failed: Seek: Track 0 not found
Error: ! Unable to open the input file: floppy.ima for reading
```

But on a SONY 2x drive without GW:
```
~$ sudo fatcat -l / /dev/sde 
Listing path /
Directory cluster: 0
f 18/9/1990 10:04:32  FRAESEN.~                                          c=2 s=608 (608B)
f 18/9/1990 10:07:52  BOHREN                                             c=3 s=5166 (5.04492K)
f 27/9/1990 15:22:10  FRAESEN.SM3                                        c=9 s=387 (387B)
f 27/9/1990 15:20:08  FRAESEN                                            c=10 s=754 (754B)
f 18/9/1990 09:58:40  LABEL                                              c=17 s=487 (487B)
f 18/9/1990 09:58:42  DURCHMES.BOR                                       c=18 s=44 (44B)
f 18/9/1990 09:58:42  DURCHMES.FRA                                       c=19 s=13 (13B)
f 18/9/1990 09:59:04  BOHREN.SM3                                         c=20 s=6281 (6.13379K)
f 28/9/1990 07:45:18  ADAPTER                                            c=11 s=5301 (5.17676K)
f 28/10/1996 09:57:06  578-STO.GBX                                        c=27 s=14811 (14.4639K)
```

xdftool doesn't work on block devices.
```
gerhard@ubuntu-SSD1:~$ sudo xdftool -f /dev/sde list
'list' IOError: can't detect type of image file
```

Keir, thank you for your generous RMA. I'm now waiting for a v4.1 exchange device.

kind regards

Gerhard Gaußling

ggaussling, 2025-03-08:
Hello Keir,

thank you very much for the GW v4.1!

In a first attempt today I thought it might not work. But I had a power only USB cable.
The second try also didn't work, but it was detected.
Therefore I thought it might be due to an defect floppy drive. And indeed. After I exchanged the drive it now works flawlessy.

I then switched the device back to V4 with the new LD1117STR and it worked too, despite the missing capacitor. I also bought a set of 0805 resistors and a capacitor kit. I'll solder them to the V4. 

So thank you very much with your hints too.

Maybe it was due t a short circuit, because the drive was affected too.

Kind regards

Gerhard Gaußling