keirf/greaseweazle #529: Feature request: SCP image good-track/sector stacking/splicing

philpem, 2025-03-07:
On occasion when imaging a ropey disc, I'll get bad sectors (or tracks), but they'll be in different places.

For example's sake, let's say the first read has the last 10 tracks bad, the second has the first three bad, and the third has a couple of bad tracks in the middle. Across all three images, I have all the data -- but spread across three SCP files.

It would be useful to have a way to automatically or manually "stack" several SCP images, find the best tracks and produce a "clean" SCP file. If the output is a cooked sector image (e.g. IMD) this could work on sector level, taking the best copy of each sector.

This has several benefits - not just the time saving but also reducing wear on already marginal media.

wohali, 2025-03-07:
Indeed, Applesauce has the ability to pause between reads, let you clean media/the drive, then restart a read (especially a cooked one) where it will retry bad sectors until it gets them (or retries time out or you cancel the operation).

Booyaka101, 2026-09-12:
I wrote a tool that does the sector-level half of this: https://github.com/Booyaka101/diskstack (`pip install diskstack`).

It takes two or more dumps of the same disk in any mix of .scp, KryoFlux .raw, .hfe and sector images, and merges per sector rather than per track, since the bad parts of one read usually aren't bad in the same places as the next. Output is a sector image plus a JSON report.

```
$ diskstack disk_a.scp disk_b.scp -o merged.img

720 sectors: 714 clean (from 2 files), 6 unresolved

Tracks needing attention
  cyl  head  clean  voted  unresolved  missing  sector ids still bad
   17     0      7      0           2        0  4,7
   18     0      7      0           2        0  4,7
   19     0      7      0           2        0  4,7

Re-read just these tracks, then run diskstack again with the new capture added:
  gw read retry.scp --tracks=c=17-19:h=0
```

The retry line is the part that matters for the wear point you raised. Three tracks on the next pass instead of eighty, and that partial capture stacks straight in with the full ones.

Attempts are grouped by the (cyl, head, sector id) in the ID address mark rather than by where they sit in the flux, because two captures of one disk don't line up, different index phase and different motor speed. A sector whose own CRC passes is taken as is. Where nothing passed, it recombines the attempts a few ways, byte-wise majority across all of them, one payload per source file, and each read on its own, and keeps the first result that satisfies a check value that came off the disk. The report names which route rebuilt each sector, so you can see what it decided rather than take its word for it.

You mentioned determinism. It vendors the gw decoders pinned to one commit, so upgrading your gw install doesn't move the output from under you.

What it won't do is your first option, a clean SCP out. Flux goes in and never comes out. The decisions happen at sector level, and splicing them back into a flux stream means writing flux that was never on the disk, so track-level SCP splicing is honestly a different tool. No manual mode either, it's all automatic, though the per-sector provenance in the report is there to check it against.

Should say up front that the damage in my fixtures is synthetic. Real flux and your real decoders, each capture given its own index phase and motor speed, and three damaged ones reconstruct a 360K image byte for byte, but I haven't had it in front of a genuinely rotting disk yet. If you still have the three SCPs that prompted this I'd be interested in what it makes of them.