keirf/greaseweazle #530: Many 3.5 inch PC drives do not work

mvbhz, 2025-03-11:
I have two versions of Greaseweazle - F1 and a newer one from the Amiga Store. However, when I tried to match them to a floppy drive, over 90% of the drives do not work - they throw errors when reading or writing. The only 2 drives that work are Panasonic and Citizen. What does it depend on?

keirf, 2025-03-11:
Are the failed drives working elsewhere?

keirf, 2025-03-11:
Could be a power issue. Try external power. Or if not tested in an independent setup the drives could simply be bad.

mvbhz, 2025-03-11:
No. It's impossible for 10 stations to be damaged. This clearly doesn't work with Greaseweazle. It works on PC.

mvbhz, 2025-03-11:
DD. I also tried on DD disks and it doesn't work either. Taping the HD hole in HD doesn't solve the problem either.

mvbhz, 2025-03-11:
The jumper is in the same place it was "from the factory". I didn't change anything. The cable is straight, not twisted. The cable is not damaged. The power cable is also not damaged.

As for the external power supply, I would have to check the separate power supply. However, I doubt that over 90% of the floppy drives I have require an external (more powerful) power source.

That's not it. I have a ready Greaseweazle device with a Panasonic floppy drive that works properly. I disconnect this floppy drive and connect, for example, an NEC floppy drive. And I get the errors as above. If it was a bad Greaseweazle configuration, it wouldn't work on the Panasonic floppy drive either.

keirf, 2025-03-11:
No idea then, sorry.

closed 2025-03-11 by keirf

mvbhz, 2025-05-13:
And what kind of cable should connect the GW to the floppy disk drive from the PC? I have the so-called "twisted". The twisted end is connected to the floppy drive and the other to the GW. Is this correct?