keirf/greaseweazle #532: Convert to HFE: Weak sectors not detected

philpem, 2025-03-18:
This is a bit of a "well HxCFE can do it..." kind of bug-but-maybe-feature-request issue.

I've been imaging some Sherston educational games for the Acorn Archimedes, initially as SCP but I want to convert them to HFE to run on an emulator (or on a physical machine with Flashfloppy). The last sector of track-0 head-0 contains a DC-erased weak sector with a valid sector header, which is used for copy protection.

When I exported the HFE from the SCP, I specified "--format acorn.adfs.800", the bitrate was picked up correctly, and the weak sector was identified as a bad sector.

Unfortunately the resulting HFE file has turned the weak sector into a valid sector (even though it said it had a bad CRC), and the protection check is now failing as a result.

Could it detect the weak sector and mark that in the HFE file?

keirf, 2025-03-18:
Does the protection require only bad CRC rather than weak bits?

philpem, 2025-03-18:
No, it's the Sherston disk protection so it needs the weak bits.

It reads the weak sector a couple of times and expects all the reads to be different. There's a valid sector address header (IDAM) and data header (DAM) but the sector data is DC-erased so reads as noise.

The CRC passing or failing is ignored by the protection code.

keirf, 2025-03-18:
You will probably find that gw convert to HFE with no format specifier works fine. You'll need to specify bitrate something like `gw convert name.scp name.hfe:: bitrate=250`

closed 2025-04-19 by keirf