keirf/greaseweazle #542: GW 4.1 with GUI: Failed to verify Track 0.1

Bersa81, 2025-04-11:
Good afternoon,

I recently bought a GW 4.1 and I am trying to figure it out.

My setup:
- Alienware laptop
- connected with USB-C cable
- powering via USB-C
- Y-E data floppy drive
- twisted cable

- Tools/FluffyFloppy
--------------------------------------------------------------------------------------------------------
Microsoft Windows [Version 10.0.26100.3476]
(c) Microsoft Corporation. All rights reserved.

C:\Users\lucab\OneDrive\Documenti\Amiga\FluxMyFluffyFloppy_5.2.4_win32bit>"C:\Users
yFluffyFloppy_5.2.4_win32bit\greaseweazle\gw.exe" info
Host Tools: 1.22
Device:
  Port:     COM3
  Model:    Greaseweazle V4.1
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.6
  Serial:   GWB021AB265976C010078D1705
  USB:      Full Speed (12 Mbit/s), 128kB Buffer

C:\Users\lucab\OneDrive\Documenti\Amiga\FluxMyFluffyFloppy_5.2.4_win32bit>
----------------------------------------------------------------------------------------------------------

- Seems that I can read disks (can read a disk working on a A500)
- I can Erase a disk
----------------------------------------------------------------------------------------------------------
T80.0: Erasing Track
T80.1: Erasing Track
T81.0: Erasing Track
T81.1: Erasing Track

C:\Users\lucab\OneDrive\Documenti\Amiga\FluxMyFluffyFloppy_5.2.4_win32bit>
----------------------------------------------------------------------------------------------------------

- If I try to write a disk with verification I get this issue:
----------------------------------------------------------------------------------------------------------
Microsoft Windows [Version 10.0.26100.3476]
(c) Microsoft Corporation. All rights reserved.

C:\Users\lucab\OneDrive\Documenti\Amiga\FluxMyFluffyFloppy_5.2.4_win32bit>"C:\Users
yFluffyFloppy_5.2.4_win32bit\greaseweazle\gw.exe" write --format amiga.amigados --
umenti\Amiga\ROMs\Bubble Bobble (1987)(Firebird).adf"
Format amiga.amigados
Writing c=0-79:h=0-1
T0.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T0.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T0.1: Writing Track (Verify Failure: Retry #1)
** FATAL ERROR:
Failed to verify Track 0.1

C:\Users\lucab\OneDrive\Documenti\Amiga\FluxMyFluffyFloppy_5.2.4_win32bit>
----------------------------------------------------------------------------------------------------------

- I was able to write ONCE the same ADF to the same disk, and it was loading into the A500
- If I try to write the ADF without verification, read it, and then load the image into HxC, the analysis shows no bad sectors
- Seems that writing a floppy without a verification leads to a non working disk (the A500 gives an error while reading)

[image]
[image]
[image]

- Tried with different floppy disks with the same result
- Sometimes you could not "feel" (no noise, no head moving) the drive unit trying to do anything while trying to write, erasing always spins the disk

Can you please help? Could it be an hardware failure from the drive? I was just very unlucky with 5 different floppies?

keirf, 2025-04-12:
Probably a drive problem. It may be that the second head cannot write. If your disks were already Amiga format then you could be reading back old Amiga data on subsequent reads.

If you erase a disk, and then read it, is the HxC view fully erased (will look all red)?

Bersa81, 2025-04-12:
Hello, thank you for your reply. I did some more test:

- Read a disk
- Check the analysis (green) 
- Format it
- Read it again
- Check it again in HxC

The analysis show it as still completely green. I was able to - or think that I was - write a disk once. 

 Can say the disk was not erased and the drive is not able to write? Repeated the test with 2 disks.

Thank you

Attached files:
- GW_Issue_3 - erase operation
- GW_Issue_4 - read after erase
- GW_Issue_5 - disk analysis in HxC after the erase

[image]
[image]
[image]

keirf, 2025-04-13:
Right, if it's still completely green after erase then writes aren't working. It could be the drive, cable, or Greaseweazle. Probably in descending order of likelihood.

Bersa81, 2025-04-13:
Thank you! I will try to change the drive and I will get back with the results.

keirf, 2025-04-19:
Sounds like you'll be fine. Please re-open if you do have issues upon further testing.

closed 2025-04-19 by keirf