keirf/greaseweazle #543: Failure to verify track 48 on Mac Formats

all2coolnick, 2025-04-12:
Can anyone suggest why my new Greaseweazle always fails at track 48 (fails to verify) when writing 400k or 800k Mac format 3.5” floppy disks? I bought a pre-built GW v4.1 board from Amigadeals in the UK. I bought a brand new Samsung floppy drive (SFD-321B) and am using a straight-wired 34pin ribbon cable (I’m specifying drive as 1 in the write command). I’ve tried various different disk images from Macintosh repository. 
I’m using latest GW software from Terminal in Mac OS 12. 
I have read that track 48 is where the drive changes speed for Mac disks and seen suggestions that some floppy drives can’t do this. I’m using brand new TDK DSDD disks from a sealed box. 
Any suggestions would be hugely appreciated.

keirf, 2025-04-13:
`I have read that track 48 is where the drive changes speed for Mac disks and seen suggestions that some floppy drives can’t do this.`

Bingo.

all2coolnick, 2025-04-13:
So how do I obtain a suitable drive? Is there any list of suitable models?
Since the creation of Mac disks is an advertised capability of the Greaseweazle, why does the official documentation say “You can connect your V4.1 to any floppy drive implementing a Shugart interface”?

keirf, 2025-04-13:
You can connect your V4.1 to any floppy drive supporting a Shugart interface. But it's drive dependent what a drive can read/write. Unfortunately it is not within the Greaseweazle's powers to perform miracles.

keirf, 2025-04-13:
I don't know where you are based but if you bought from me in the UK you are also welcome to return the board for refund. Most other resellers will also oblige.

keirf, 2025-04-13:
I have updated the wiki pages to clarify the position on GCR formats (Apple II, Mac, C64). The Yann Serra tutorial was particularly gung ho!

all2coolnick, 2025-04-13:
Thanks for the replies. So if I understand you correctly, you’re saying that although pretty much all PC floppy drives use the Shugart/IDE 34-pin interface, some don’t fully implement that interfaces speed control function? 
It would be a shame to return my Greaseweazle since what it potentially offers is so useful. I’m pretty sure it is used by many Apple enthusiasts to produce Mac 400K and 800K disks so it would be great if there was a way for the community of users to add to a list of drive models that support this ability.

keirf, 2025-04-13:
Rotational speed control is very specific to proprietary Mac drives. Actually when reading from a normal drive the speed remains the same and the bitrate varies. It's this varying that makes Mac disks particularly problematic. Even more so than Apple II and C64 disks. It can also be affected by floppy media stock -- especially HD disks with the HD hole covered basically never work (but they are pretty reliable for MFM disk formats), and it might well be the case that even DS stock can be problematic (but that's unproven).

Regarding drives, I picked two randomly off my pile: NEC FD1231M got further than track 48 but with patchy reliability and eventually gave up around track 65. TEAC FD-235HF (a venerable older model of drive) performed flawlessly.

keirf, 2025-04-13:
I also have a Samsung SFD-321B and that performed flawlessly too. So I suspect your DD disks are causing trouble here for some reason. Ironically I tested with an old used unbranded DD disk from the Amiga era.

all2coolnick, 2025-04-13:
Thanks so much for looking into this. It’s so helpful having others experiences. I have purchased new, sealed DSDD disks from two eBay sellers. When I had problems with the first batch, I noticed patchiness on the disk surface so sent them back. The new floppy drive had also got a bit battered in transit so I sent that back and got a replacement. Still no joy. The second batch of disks look perfect but still failing. 
Could it be to do with either the ribbon cable I’m using (straight wired) or the power being supplied to the drive through the Greaseweazle (possibly not having enough juice for the speed change)?
As it happens, I recently purchased a large batch of used DSDD Amiga disks so I will try some of those.

keirf, 2025-04-13:
Try a few Amiga disks after checking them for cleanliness. Hard failure at track 48 is in my limited experience media rather than drive. However just because we have the same Samsung drive model also does not mean they are identical -- there can be big changes between revisions.

keirf, 2025-04-19:
Any luck?

all2coolnick, 2025-04-19:
Thanks so much for following up. I’ve been unwell all this week so haven’t had a chance to try yet but I will. I had another, older 3.5” floppy drive arrive which I will check too. NEC FD1331M. 
I’ll let you know how I get on.

dekkit, 2025-04-20:
> Can anyone suggest why my new Greaseweazle always fails at track 48 (fails to verify) when writing 400k or 800k Mac format 3.5” floppy disks? I bought a pre-built GW v4.1 board from Amigadeals in [...]

I've recently had success with writing 800KB Mac floppies with Greaseweazle and a standard Fixed speed 3.5inch IBM HD floppy  drive and using new old stock double density disks that were IBM formatted out of the box.     I typed my notes based on what found on various forums and from my own experiences in the following issue - see https://github.com/keirf/greaseweazle/issues/538

A few things to note that hopefully assists:
- You cannot write Mac 400KB / 800KB disk images 'as is' without further adjustment / conversion. 
- I've typed up further background as why to this is needed (see issue linked) based on what i found / discovered on various forums (when i started on this journey a few weeks ago).  
- In short, you need to spread out the raw flux transitions contained in the disk image via software so they can be written by a fixed speed IBM drive (i.e. recalculate the position of data).
- Linked in the issue is a tool that will perform this conversion - it will convert the image to individual flux transitions so the data will be written in a way that the a variable speed drive will recognize.  
- the other thing to check, is your Greaseweazle settings.   Make sure you have not accidentally used the "double-step=2" switch (you do not need it for double density 720KB floppy disks in HD drives).   I made this mistake and i would get an error sometime after the ~40 track mark regardless of the disk used.  3.5inch HD drives can write double density disks without the need for this switch.   This switch is only relevant to 5.25inch HD floppy drives (as i subsequently discovered)!.

All the best with it.

keirf, 2025-04-20:
Another option there, although gw is able to do the format conversion to raw on the fly `--format=mac.800` for example.

all2coolnick, 2025-04-20:
Thanks dekkit for sharing your experiences.

dekkit, 2025-04-20:
> Another option there, although gw is able to do the format conversion to raw on the fly `--format=mac.800` for example.

Oh even better!  Using the --format=mac.800 avoid the need to use that extra conversion tool.    Have added that to my notes and will update the issue - thanks for the info.

all2coolnick, 2025-04-20:
OK, that’s confused me a bit. If the mac.800 format parameter (which I am using) causes the .img data to be re-written to take account of, and remove the need for, the speed change, then why am I getting the track 48 error? I thought that was due to the drive needing to change speed at that point, and that the use of a shugart interface drive was so that the host could directly control aspects like the rotational speed. Sorry if I’m being dense.

keirf, 2025-04-20:
Most drives are fixed speed. So the timing of bits being written to the disk must change instead. That's what the mac.800 parameter does. Concretely: A Mac drive slows down rotationally as it moves to inner tracks; But it is equivalent to rotating at a constant speed and decreasing the data rate (ie. spacing out more the timing of flux changes)

all2coolnick, 2025-04-20:
OK, thanks. So if the drive does not need to change speed while writing because the data has already been pre-processed by GW to compensate, why is track 48 problematic?

keirf, 2025-04-20:
Because the drive electronics are designed for MFM  bit timings and they start to amplify noise if the bit timings get further apart than expected. This effect is drive and media dependent, and mainly on the read side (I believe). That is you may find that if you write a disk without verification beyond track 48, it may work on a Mac drive.

all2coolnick, 2025-04-20:
Lots to think about and try. Thanks for taking the time to explain.

all2coolnick, 2025-04-22:
> ...you may find that if you write a disk without verification beyond track 48, it may work on a Mac drive.

I'm not having any better luck with the used Amiga DD disks so wanted to try your suggestion about only verifying up to track 47. I can't find any mention in the GW documentation of a command parameter to turn verification off. Can you point me in the right direction? Ideally it would be great to be able to specify the verification for a range of tracks but presumably if this is not possible, I could write the disk in two steps with the second command having verify off and specifying track from 48 onwards.

keirf, 2025-04-22:
Do `--tracks=c=0-47` and then `--tracks=c=48-79 --no-verify`

dekkit, 2025-05-02:
> I'm not having any better luck with the used Amiga DD disks so wanted to try your suggestion about only verifying up to track 47. I can't find any mention in the GW documentation of a command [...]

I needed to do some more Mac.800 disks for a friend tonight and he gave me a box of ~10 'used' disks that have been formatted in a Mac drive.

All but 1 failed at various track positions no matter how many repeated attempts, so i then resorted to using a few more of the new original double density stock.

In short - using 'used' disks caused me quite a few issues, which may also be part of your issue (almost all had issues with tracks) 

A few more insights when using 'used' DD disks :

gw command I used...
`gw.exe write --diskdefs "C:\Users\..\Downloads\greaseweazle-1.16.3-win64\greaseweazle-1.16.3\diskdefs.cfg" --format=mac.800 --retries=3 --drive=A "C:\Users\..\Downloads\greaseweazle-1.16.3-win64\greaseweazle-1.16.3\Input\System Tools.img"Writing c=0-79:h=0-1`

**Results with 'used' DD disks**
- I saw the `** FATAL ERROR: Failed to verify Track 49.0 `on one disk and `** FATAL ERROR: Failed to verify Track 48.0` on another - which i thought i may have reproduced your problem.
- Then i tried a few more times on few more disks until I eventually landed on` "All tracks verified"`.    Looking through the log it contained a few verify lines like "...`(Verify Failure: Retry # 1) `indicating that the verify feature does work and it was progressing with my fairly standard IBM 1.44MB drive.
- When I was previously using the other method (using the macintosh conversion app), I came to realise that when I later wrote the raw flux tracks that this approach does not verify the data at all.   It provided a message at the end: `"No Tracks verified (Reason: Verify unavailable)"`.
- In reflection using the other method was bit hit and miss, as I needed to test it on the real mac to see if booted / worked at all (the results for the 2 x disks = 1x disk booted and seem to run fine and  1 x disk did not).
- Using the switch `--format=mac.800` is definitely the better option given this verification step.
- For the 1 x used disk that I got working, I eventually split the tracks  using the switch something like `--tracks=c=48-79` to retry quite a few times on problem track.
- Turning off the verification as in the above reply with the switch ` --no-verify` may at least get you booting and skip any bad tracks.

When i ended up resorting to using new DD disks (e.g. best case test scenario), I thought i'd capture my notes here as well when using the  switch `--format=mac.800`

gw command I used... (same as above)
`gw.exe write --diskdefs "C:\Users\..\Downloads\greaseweazle-1.16.3-win64\greaseweazle-1.16.3\diskdefs.cfg" --format=mac.800 --retries=3 --drive=A "C:\Users\..\Downloads\greaseweazle-1.16.3-win64\greaseweazle-1.16.3\Input\System Tools.img"Writing c=0-79:h=0-1`

**Results with 'new old stock' DD disks**

- Even with new disks i would sometimes get a number of retries
```
T65.1: Writing Track (Verify Failure: Retry #1)
T65.1: Writing Track (Verify Failure: Retry #2)
T65.1: Writing Track (Verify Failure: Retry #3)
```
- I ended resorting to increasing the Retries to 5 (even 8 retries at one stage) -  to ensure the track was confirmed written.
- I was able to get the `All tracks verified`
- All disks were later tested as working successfully in a real Mac.

Here is a full log of one successful disk being written (note: I had set retries = 8 by the this stage) - in case it helps

```
Writing c=0-79:h=0-1
Format mac.800
T0.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T0.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T1.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T1.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T2.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T2.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T3.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T3.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T4.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T4.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T5.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T5.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T6.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T6.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T7.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T7.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T8.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T8.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T9.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T9.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T10.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T10.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T11.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T11.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
[... cut ...]
T77.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T77.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T78.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T78.0: Writing Track (Verify Failure: Retry #1)
T78.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T79.0: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
T79.1: Writing Track (Flux: 200.0ms period, 219.9 ms total, Terminate at index)
All tracks verified
```