keirf/greaseweazle #544: How to verify or format floppy?

mersenne-twister, 2025-04-13:
I'm wondering how I can go about verifying a floppy disk, without explicitly writing anything in particular to it.

I'm also curious about if it is possible to format a floppy with an empty filesystem for a given disk type.

keirf, 2025-04-13:
1. Write a disk image to it. Eg. 720k PC IMG for DSDD, 1.44M PC IMG for DSHD, etc. This will write+verify, and thus verify the media.
2. Greaseweazle tools don't currently do filesystems. You would need an empty filesystem disk image, and write that.

Other solutions for (2). FluxEngine I think has a GUI, can talk filesystems, and can talk to Greaseweazle; And on Windows there is Rob Smith's DiskFlashBack which can read/write many filesystems, and format them, to disk image or direct to Greaseweazle.

mersenne-twister, 2025-04-15:
> 1. Write a disk image to it. Eg. 720k PC IMG for DSDD, 1.44M PC IMG for DSHD, etc. This will write+verify, and thus verify the media.
> 2. Greaseweazle tools don't currently do filesystems. You would [...]

Thank you! I've been using flux engine, and found it to be effective for what I'm doing.

closed 2025-04-15 by mersenne-twister