keirf/greaseweazle #551: "gw convert foo.scp foo.dsd" creates 200kB DSD, BeebJit (correctly) creates 400kB DSD from Acorn 2-sided DFS disc

evoncken, 2025-05-02:
Hi,

First and foremost, thanks for creating GreaseWeazle 4.1 and FlashFloppy ;-)

I'm running into an issue with SCP to Acorn Dual-Sided Disk (DSD) conversion.

GW host tools 1.22 on Win11, command line:
      gw read --drive A --revs 5 --tracks c=0-79:h=0-1 diskname.scp

Note that source floppy appears to be formatted both sides, but only one side contains files.

Conversion using Greaseweazle host tools:
    gw convert diskname.scp diskname.dsd (200kB file)

This yields a DSD file that, when loaded into BeebEm, shows plausible content for "*CAT 0", but shows corrupt data for "*CAT 2":
[image]

And same SCP image, but converted with beebjit:
    beebjit -0 diskname.scp -convert-ssd (400kB file)

This shows an empty disk, as expected:
[image]

I suspect this might be a bug in the "gw convert" command. I can provide the SCP file (62MB uncompressed, 19MB compressed) if needed - seems a bit large to attach to this issue?

evoncken, 2025-05-02:
Forgot to add details:

```
Host Tools: 1.22
Device:
  Port:     COM5
  Model:    Greaseweazle V4.1
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.6
  Serial:   GWB01431455976C01007011705
  USB:      Full Speed (12 Mbit/s), 128kB Buffer
```

keirf, 2025-05-02:
Yes please attach the SCP and the Beebjit-generated image, zipped up. GitHub should allow up to 25MB attachments, and who are we to argue.

evoncken, 2025-05-02:
[music500-dsdd.zip](https://github.com/user-attachments/files/20013553/music500-dsdd.zip)

[music500-dsdd-beebjit.scp.zip](https://github.com/user-attachments/files/20013556/music500-dsdd-beebjit.scp.zip)

[music500-dsdd-gw.zip](https://github.com/user-attachments/files/20013562/music500-dsdd-gw.zip)

evoncken, 2025-05-02:
For completeness, adding the actual batchfile I used to read/convert the floppy

[DS80-read.zip](https://github.com/user-attachments/files/20013620/DS80-read.zip)

keirf, 2025-05-02:
Okay, there is now an automated build of the tools you can try (or you can pipx install, or similar, from source if you like). The command line you need is:
```
gw convert --format acorn.dfs.ds80 diskname.scp diskname.dsd
```

And you should see no warnings, and the final sector map should be clean (all `.` and no `X`).

keirf, 2025-05-02:
Automated builds for Windows (you need to be logged in to see the download links): https://github.com/keirf/greaseweazle/actions/runs/14796541349

evoncken, 2025-05-02:
Wow! You probably won the world record of committing a bugfix - 5 minutes or less ;-) THANK YOU!

evoncken, 2025-05-02:
Running version 1.23dev8 - the SCP file is still converted to 200kB DSD, but a *CAT 2 now correctly shows an empty directory listing = yay! ;-)

keirf, 2025-05-02:
It will be converted to a 400kB file if you use the correct format: `--format acorn.dfs.ds80` 

Note the 80 on the end.