keirf/greaseweazle #565: output error sector data from flux files instead of generic filler -=[BAD SECTOR]=-

hex-dreamer, 2025-07-09:
Greaseweazle "gw convert" does (in my experimentation) the best job of converting a raw flux file to sector data.

My request below may be simply a case of me needing to do more reading about the command line options - so my apologies for taking your time up in advance - in that case just tell me to RTFM.

REASON for this request ...
 I have often found that I can reconstruct most (sometimes all) the bad sector data from different bad readings of the same sector.

(Warning Pseudo code below...)

For faulty fragile disks , I first 

 "gw read"   KryoFlux.raw   revs=3
 
I then 

 "gw convert" KryoFlux.raw  --format.ibm360  diskimage.img 

Obviously "diskimage.img " is filled generic -=[BAD SECTOR]=- filler for bad sectors

My ideal scenario ----

gw convert  --format.ibm360 --RAW revs=1   diskimage.IncludingRawBadSectorsRev1.img
gw convert  --format.ibm360 --RAW revs=2   diskimage.IncludingRawBadSectorsRev2.img
gw convert  --format.ibm360 --RAW revs=3   diskimage.IncludingRawBadSectorsRev3.img

I can't figure out how to get access to those individual readings from each revolution, and I don't want to keep using the fragile disks when I have a very good kryoflux.raw file containing 

I am aware "gw convert" will read all revs and select the best valid sectors, and that is fine for valid sectors.

I am also aware of the --raw option (like below) , but that means I have to read the disk again. 
 gw read --format=ibm.720 --raw really_raw_flux.scp

keirf, 2025-07-09:
So I'm unclear what you need here. If there is at least one good read of a sector in a multi-revolution dump, then that will be extracted into the destination IMG file already. Or are you concerned about partially-good sectors and different handling for those?

boamaod, 2025-08-03:
I think this has been already explained and proposed in #273, similar discussion can be also found in #373. Currently incorrect CRC seems to entail omitting actually saved data altogether. This might be good default for well preserved data, but some low quality data, especially text can be read or corrected despite any CRC errors. I tend to agree with @keirf, that retaining the original data with incorrect CRC is "probably better than trashing the data completely, and writing a good CRC".