keirf/greaseweazle #568: Sector length issues when writing .IMA/.IMG with ibm.1440

lnxsrt, 2025-07-20:
Recently got my GW 4.1 up and running. I'm having a strange issue when writing .IMA/.IMG files in the ibm.1440 format. It seems like the physical sector length or gap between sectors is off? There is a large gap present between the first and last sectors after writing a disk using GW. The disk will work in other drives, but seems to struggle to read at times.

`gw write --format=ibm.1440 --drive=B Ghoflp5.ima`

`gw read gho_bad.scp --drive=B`

[image]

The same .ima when written with WinImage

`gw read gho_good.scp --drive=B`

[image]

As you can see, no bad sectors, checksum matches, and will read in a DOS computer. There is a noticeable delay when attempting to initially read on the DOS computer.

I've also read many other disks, formatted by my drives, and some OEM, they all have a much smaller gap (like the second screenshot) between first and last sectors.

keirf, 2025-07-21:
I think it's harmless and what Greaseweazle does is closer to standard spec for 512-byte sectors (standard 84-byte GAP3 between sectors, compared with more like 108-byte GAP3 on some OEM PC disks). It's not going to make any difference for reads. The GAP3 is to prevent next sector being overwritten when writing a single sector, due to rotational speed differences between drives.

I could look at increasing GAP3 but then I can equally be dinged for being against the standard datasheets. Also you can make your own diskdef with the GAP3 of your choice.

lnxsrt, 2025-07-22:
Thanks keirf. Your custom diskdefs created disks very similar to my second image. It also fixed the read delay issue I was having on one machine. 

Doing some further thinking here, I also noticed that on the tighter formatted (default disk_defs; 84 gap) disk the top and bottom data areas are not aligned. I'm guessing this will impact filesystem read/write speeds on a real machine, but I want to do some further testing.

IMD also uses the default 84 gap, but writes disks that look the second image. My gut reaction looking at the images, was some sort of an off-by-one error, like the gaps were calculated assuming there was an extra sector (19 instead of 18) to be written. But, I also figured that bug would have been found long ago and was probably not likely :-)

lnxsrt, 2025-07-22:
I just tested read speeds (no caching or anything enabled), identical disks written with GW, one with 84 gap, other with 108 gap.

84 gap - 1 minute 10 seconds
108 gap - 1 minute 1 seconds

Roughly 14.7% slower reading on real hardware. Sounded like more seeking as well.