keirf/greaseweazle #570: ControlCmd.ClearComms causes a traceback due invalid value on ppc64le

sharkcz, 2025-07-28:
I am trying to use Greaseweazle on my ppc64le workstation running Fedora Linux 42 (kernel-6.16.0-0.rc6.52.fc43.ppc64le) and I am getting a `FATAL ERROR` and/or a traceback (with `--bt`) during the startup.
````
[dan@talos pc]$ gw --bt info
Host Tools: 1.22
Device:
Traceback (most recent call last):
  File "/usr/bin/gw", line 8, in <module>
    sys.exit(main())
             ~~~~^^
  File "/usr/lib64/python3.13/site-packages/greaseweazle/cli.py", line 69, in main
    res = main(argv)
  File "/usr/lib64/python3.13/site-packages/greaseweazle/tools/info.py", line 73, in main
    usb = util.usb_open(args.device, mode_check=False)
  File "/usr/lib64/python3.13/site-packages/greaseweazle/tools/util.py", line 476, in usb_open
    usb = USB.Unit(serial.Serial(devicename))
  File "/usr/lib64/python3.13/site-packages/greaseweazle/usb.py", line 194, in __init__
    self.reset()
    ~~~~~~~~~~^^
  File "/usr/lib64/python3.13/site-packages/greaseweazle/usb.py", line 223, in reset
    self.ser.baudrate = ControlCmd.ClearComms
    ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.13/site-packages/serial/serialutil.py", line 299, in baudrate
    self._reconfigure_port()
    ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/lib/python3.13/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
    ~~~~~~~~~~~~~~~~~^
        self.fd,
        ^^^^^^^^
        termios.TCSANOW,
        ^^^^^^^^^^^^^^^^
        [iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
termios.error: (22, 'Invalid argument')
````
Seems the ppc64le kernel doesn't allow arbitrary baud rates.

sharkcz, 2025-07-28:
FWIW changing `ControlCmd.ClearComms` to 19200 "unblocks" the startup
````
[dan@talos pc]$ gw --bt info                                                                                                                                                             
Host Tools: 1.22
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4.1
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.5
  Serial:   GWB014313A5976C010071C1705
  USB:      Full Speed (12 Mbit/s), 128kB Buffer

*** New firmware version 1.6 is available
To perform an Update:
 - Run "gw update" to download and install latest firmware
````

keirf, 2025-08-04:
This doesn't appear to even be specific to ppc64le.

keirf, 2025-08-04:
Yeah I don't know exactly where the error is coming from, it may not be kernel:
* Python `pyserial` and `termios` packages
* glibc or other low-level runtime library
* Kernel

keirf, 2025-08-04:
If there's a conflation of issues here, and yours really is a longer standing ppc64 issue, that's quite a coincidence. Not impossible though!

sharkcz, 2025-08-04:
I have run `gw` under strace, so the error is coming from the kernel (`ioctl()` syscall failing), but the root cause doesn't have to there. The test from https://github.com/GrantEdwards/Linux-arbitrary-baud (after little edits for ppc64le) seems to run OK. So the suspects are likely the Python `pyserial` and/or `termios` packages ...

I think my other question is, why the 10000 speed is part of the reset procedure? Could it be achieved in some other/safer way?

keirf, 2025-08-04:
My guess is Python not using the termios2 API, which is supposed to be used for arbitrary baud setting AIUI. So another option is to fix that, or hack around it in Greaseweazle with something raw and Linux specific.

keirf, 2025-08-04:
> I suspect it fails somewhere inside the Python `termios` module, because I can't find a failed `TCSETS` `ioctl()` call or similar in the strace output ... Seems I was wrong about it in my previous comment.

Yeah that's why I'm a kernel hacker; I hate middleware/runtime.

keirf, 2025-08-04:
It should be possible to download and build the pyserial package locally, and also pull the termios module out of cpython and build that too. Could even build a whole cpython, install locally, and pip install a local version of pyserial into that environment. Then have at pyserial and termios with tracing etc. 

The troubling thing is I can't see any scary recent changes to pyserial or termios.

keirf, 2025-08-05:
That at least looks like a sensible path by which to set a custom baudrate, using the termios2 API. Does the Linux-arbitrary-baud test still work on Fedora Rawhide? That uses TCGETS2 ...

keirf, 2025-08-05:
If you verify glibc is the problem, the simplest solution might be to directly invoke the ioctl from Greaseweazle. Unfortunately that's probably not portable across architectures, even the syscall number can vary. We could raise a bug against glibc though!

wohali, 2025-08-06:
Hey Keir, https://sourceware.org/pipermail/libc-alpha/2025-July/168420.html is the most likely place to start looking. Note the complete removal of `TCSETS` on Linux and a full removal of those ioctls.

keirf, 2025-08-06:
Perhaps it's as simple as an incorrect header file inclusion in Python's termios then?

dennisbusch-de, 2025-08-14:
Don't know if this is helpful information but downgrading *glibc* on my Arch system fixes the issue for me, so the problem appears to be indeed related to recent changes in that library.

```
downgraded glibc (2.42+r3+gbc13db739377-1 -> 2.41+r65+ge7c419a29575-1)
downgraded lib32-glibc (2.42+r3+gbc13db739377-1 -> 2.41+r65+ge7c419a29575-1)
```

kallisti5, 2025-08-25:
See https://github.com/pyserial/pyserial/issues/805  for this issue and a fix.

pyserial sadly looks unmaintained currently.... which is going to be a whole thing it feels like 😓

kallisti5, 2025-08-25:
I submitted a patch to pyserial based on the recommendations in issue 805 on their project:

https://github.com/pyserial/pyserial/pull/808

I can confirm that change fixes the issue on the latest Arch Linux:
```
$ uname -a
Linux eris 6.16.2-2-cachyos #1 SMP PREEMPT_DYNAMIC Fri, 22 Aug 2025 16:13:15 +0000 x86_64 GNU/Linux
[kallisti5@eris ~]$ gw info
Host Tools: 1.22
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4.1
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.6
  Serial:   GWB049C8485976C01007A49705
  USB:      Full Speed (12 Mbit/s), 128kB Buffer
```

I haven't tested functionality yet though... waiting on a floppy ribbon cable for my GreaseWeazle

lnxsrt, 2025-09-13:
pyserial looks completely dead, unfortunately.

lnxsrt, 2025-10-19:
Bizarrely, greaseweazle just started working again for me on Arch. The python-pyserial package is still from Dec 2024, so that was not updated. I'm guessing the update from glibc 2.42+r3+gbc13db739377-1 -> 2.42+r17+gd7274d718e6f-1 fixed it based on the post by @dennisbusch-de 

Anyone else finding it working again?

sharkcz, 2025-11-03:
Based on the PRs I am seeing for other projects (https://github.com/wvdakker/gtkterm/pull/82, https://github.com/sigrokproject/libserialport/pull/24) I suspect there is work to be done for pyserial as well ...

dr-waterstorm, 2025-11-06:
@lnxsrt I can confirm that it started working again on my Arch install.