keirf/greaseweazle #573: Fatal error on Archlinux after update

dr-waterstorm, 2025-08-03:
First I want to thank you for this great project. It's a really nice and handy tool. 

I've been using it to archive old floppies and it worked just fine ~2 weeks ago.
Today, when I was trying to continue archiving the floppies, after updating my Arch installation to the latest version, greaseweazle fails to detect / use the drive. Instead of the usual info print with `gw info` I get:

```
Host Tools: 1.22
Device:
** FATAL ERROR:
(22, 'Invalid argument')
```

I tried it on a different computer with an Archlinux installation and both fail with the same error message (both systems are up to date).

However it does still work on my Gaming distro Nobara (Fedora) on the same computer / hardware:

```
Host Tools: 1.22
Device:
  Port:     /dev/ttyACM0
  Model:    Greaseweazle V4.1
  MCU:      AT32F403A, 216MHz, 224kB SRAM
  Firmware: 1.6
  Serial:   GWB021BAA25976C01007901705
  USB:      Full Speed (12 Mbit/s), 128kB Buffer
```

The kernel is one minor version different but it used to work just fine on Arch as well. I did not change anything on my laptop aside from updating the OS (as I rarely use my Laptop atm).

```
Linux 6.15.8-arch1-2 #1 SMP PREEMPT_DYNAMIC Tue, 29 Jul 2025 15:05:00 +0000 x86_64 GNU/Linux
```
vs.

```
Linux 6.15.7-200.nobara.fc42.x86_64 #1 SMP PREEMPT_DYNAMIC Fri Jul 18 17:26:36 UTC 2025 x86_64 GNU/Linux
```

Both are on Python 3.13.5

I'm not sure how to debug this any further. If I can be of any help by providing more feedback etc. please let me know. I'd be happy to share more.
I hope this is only a minor problem and you'll be able to fix this easily.

ArKay74, 2025-08-03:
The same on CachyOS (which is also arch-based).

`❯ gw --bt info
Host Tools: 1.22
Device:
Traceback (most recent call last):
  File "/home/arkay/.local/bin/gw", line 7, in <module>
    sys.exit(main())
             ~~~~^^
  File "/home/arkay/.local/share/pipx/venvs/greaseweazle/lib/python3.13/site-packages/greaseweazle/cli.py", line 69, in main
    res = main(argv)
  File "/home/arkay/.local/share/pipx/venvs/greaseweazle/lib/python3.13/site-packages/greaseweazle/tools/info.py", line 73, in main
    usb = util.usb_open(args.device, mode_check=False)
  File "/home/arkay/.local/share/pipx/venvs/greaseweazle/lib/python3.13/site-packages/greaseweazle/tools/util.py", line 476, in usb_open
    usb = USB.Unit(serial.Serial(devicename))
  File "/home/arkay/.local/share/pipx/venvs/greaseweazle/lib/python3.13/site-packages/greaseweazle/usb.py", line 194, in __init__
    self.reset()
    ~~~~~~~~~~^^
  File "/home/arkay/.local/share/pipx/venvs/greaseweazle/lib/python3.13/site-packages/greaseweazle/usb.py", line 223, in reset
    self.ser.baudrate = ControlCmd.ClearComms
    ^^^^^^^^^^^^^^^^^
  File "/home/arkay/.local/share/pipx/venvs/greaseweazle/lib/python3.13/site-packages/serial/serialutil.py", line 299, in baudrate
    self._reconfigure_port()
    ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/home/arkay/.local/share/pipx/venvs/greaseweazle/lib/python3.13/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
    ~~~~~~~~~~~~~~~~~^
        self.fd,
        ^^^^^^^^
        termios.TCSANOW,
        ^^^^^^^^^^^^^^^^
        [iflag, oflag, cflag, lflag, ispeed, ospeed, cc])
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
termios.error: (22, 'Invalid argument')

~
❯ 
`

keirf, 2025-08-04:
Looks like a duplicate of #570 which is not specific to ppc64le after all. Must be a regression in latest Linux kernels.